mitra42/frugal-iot #6: Actuator API

mitra42, 2024-08-29:
Top level for actuators - which is basically anything it does from blinking LED, to outputing on pins, to switching a relay

- [ ] Define actuator API 
- [ ] Review ESPHOME to see if can use
- [ ] Think through configuration issues

### Template for each actuator

- [ ] Obtain example(s)
- [ ] Build simple interface
- [ ] Consider if can redesign frugal to direct attach to pin
- [ ] Consider configuration and add to _settings.h
- [ ] Document

mitra42, 2024-09-06:
Blinken is added, and compiled, but not yet tested on a board.

janandrzejewski, 2024-09-07:
I tested on my Arduino Uno. It worked fine

mitra42, 2024-09-08:
Thanks for visiting - its just the beginning of the project, and always happy to get suggestions.

mitra42, 2024-12-22:
Closing - got a couple of examples of digital actuators now

closed 2024-12-22 by mitra42