mitra42/frugal-iot #25: Refactor: Consider reworking as classes

mitra42, 2024-10-01:
The current code is very Arduino-standard, with each .cpp having setup and loop . THis works well for now but ....

It would be worth considering reworking as a class hierarchy
Base
* Sensor
  * I2C_sensor
  * Analog_sensor
  * Digital_sensor
* Actuator
* Control

Some of the code could then reuse code fragments - so for example the loop could automatically do the nextLoopTime recording if the class had a "MS" member. 

One consideration would be saving code memory space, but another would be whether sensors  etc then became easier or harder to add to the project.

mitra42, 2024-12-22:
This has been done for the ANalog and Digital sensor - not the I2C as only have one example to work with so not sure how much is common across sensors. 
Need to do some more on the control, control_mqtt_demo is looking close, but needs generacizing #82

jksemple, 2024-12-31:
Good call. You should be able to build up a std::vector of Base objects from your configuration data, iterate through it and call Base::setup() to initialise and call Base::loop() on each one at an appropriate time interval.

mitra42, 2025-01-03:
Yes-  I was thinking of something like that, though I'd use a linked list, because a std::vector is going to need to know its content at compile time - do-able, but a linked list is as easy.

mitra42, 2025-01-03:
Here's what I'm thinking ... 
Note - there may be some auto-capitalisation by editor which I might have missed.
 
* Frugal Base
    * Sensor - reads sensors and sends as messages
        * SensorAnalog - reads a sensor via the analog pins
    * Actuator - receives messages and sets local output
        * ActuatorDigital - sets a digital pin 
    * Control - Receive a number of messages - could be local node or remote, runs calculations and sets one or more outputs via messages
        * Control3 - standard format for a basic controller, up to 3 inputs and 3 outputs
    * System - other classes for example for WiFi 
        * ??? - WiFi - manage wifi connection and portal 
        * ??? - MQTT - send and receive messages
        * ??? - Discovery - helping the UI understand what this node is

Methods & Variables

Some open questions:
* three different types of value - Bool, Int, Float and their conversion and storage. 

Converters from Sring of payload to Int, Float or Bool
* textToInt - convert text input to value reqd
* textToFloat - 
* textToBool

Frugal Base
* virtual void setup()
* virtual void loop()
* virtual dispatch() - checks if matches topic and if so calls convertAndSet
* ??? Virtual convertAndSet(String) - convert to local representation
* next - pointer to next instance in a list
* Static dispatchAll - calls system; sensor; actuator; control. dispatch
* Static setupAll - calls system; sensor; actuator; control.setupAll

Sensor
* Static *first - first sensor instance in a list
* Static setupAll - iterate thru sensor instances.calling setup()
* Static loopAll -  iterate -> loop()
* Static dispatchAll - currently null as sensors don’t receive messages
* read - read physical sensor - e.g. A2D or I2C and convert to float or int <<thought reqd>> 
* set(newvalue) - set a new value calls changed then act 
* Virtual bool changed(new value)- check if new value is valid and different from old value
* virtual act - convert value to text and send a message

Actuator
* Static *first - first actuator instance in a list
* Static setupAll - literate thru actuator instances calling setup()
* Static loopAll -  iterate -> loop()
* Virtual Set(String) - sets actuator based on text - converting 
* Act - turn value into an action - e,g. set a digital pin 

Control
* Static *first - first control instance in a list
* Static setupAll - iterate thru control instances .calling setup()
* Static loopAll -  iterate -> loop()
* More complex methods around set and act - TBD as part of Issue xx 

Control3 - subclass of Control 
* InputX_value
* InputX_topic - if set listen to that topic to get value
* InputX_control - if set, then listen to this topic to get InputX_topic
* Void process - process to look at inputs and setup outputs
* OutputX_value - the value set by a process
* OutputX_topic - where to send the value
* OutputX_control - listen for OutputX_topic

System
* Static *first - first system instance in a list
* Static setupAll - call in order: WiFi ; Mqtt; Discovery; then iterate thru other sensor instances calling setup
* Static loopAll -  iterate -> loop()

mitra42, 2025-01-03:
I've created a new branch to work on this  `24-classes`

jksemple, 2025-01-03:
> Yes- I was thinking of something like that, though I'd use a linked list, because a std::vector is going to need to know its content at compile time - do-able, but a linked list is as easy.

Not entirely sure what you mean here. You should be able to create an empty std::vector <FrugalBase> and then add elements of whatever subclass you wish as you retrieve them from your persistent storage. However using std::list will allow a configured sensor to be removed from your config more easily than std::vector so I'd go with your decision for that reason

mitra42, 2025-01-03:
OK - thanks - I mostly program in recent years in javascript, and have done C in *very* constrained (memory) environments,  and am relatively new to C++ (e.g. never used any of the template functionality) but  just took a look and didn't realize std::vector had more power than I thought.   I'm not expecting config changes (such as adding sensors) once its compiled but I *am* expecting changes to just the _local.h file to change what gets compiled. Looks like std::list or std::vector would work. I've just pushed the code that does this (list management being trivial to code) but might recode using std::list later

jksemple, 2025-01-03:
I'm only really about a year ahead of you with c++.
I was thinking you could probably set an ultimate goal to have only one compiled firmware for all devices and configure each device in the field through a device config website using pin selections and sensor/actuator choices stored in NVS using the Preferences library. I persist my collection of WiFi credentials and about 50 other settings using Preferences, (which does wear levelling behind the scenes). But for now that might be an unnecessary stretch.

jksemple, 2025-01-03:
Remember atoi() and atof() are available to convert strings to int and float respectively.  Not aware of an equivalent for converting to boolean but it's pretty trivial.

mitra42, 2025-01-03:
I originally thought about having a single compiled firmware, but the problem is memory - there is a substantial price advantage to using smaller devices, and the ESP8266 has a lot less memory than ESP32's - especially the earlier (cheaper) versions of it. In addition you don't need to use up NVS and don't need the code for substantial string handling to for example decode a JSON or YAML file or some other form of Preferences file (I'm not familiar with the "Preferences Library" you refer to above.

jksemple, 2025-01-03:
Sure. Esp8266 is cheaper and doesn't have many free gpios anyway so running more than a couple of sensors might be challenging. If you compile a bunch of firmware binaries prepared for each of the different sensors/actuators you can then do the clever config management in the OTA server where you'll have a lot more resources to throw at the problem. An installer could interact with a Web server running on your OTA server to select the required firmware for the device to replace a simple standard OTA enabled bootstrap firmware.

mitra42, 2025-01-03:
yes- that's similar to the approach I'm going for, I've got OTA working on the ESP8266's got to make one refactor before pushing it. This is currently just for updates, but will do the configuration step after.

mitra42, 2025-01-05:
I've refactored all the sensors as classes - its Branch 25 - although I've tested with all sensors, I probably won't merge it into Master until I've gone through the same process on Actuators at least.

jksemple, 2025-01-05:
> yes- that's similar to the approach I'm going for, I've got OTA working on the ESP8266's got to make one refactor before pushing it. This is currently just for updates, but will do the configuration step after.

You could perhaps include the option in your bootstrap firmware to choose which OTA firmware the installer wanted to load.  You can then send that as an extra header in your OTA request so the OTA server knows which one to send back. That field could be included on the same page as the WiFi creds as a dropdown list.

mitra42, 2025-01-06:
I like that idea it would be a relatively foolproof way to do it as the dropdown would default to whatever was installed at the time.

mitra42, 2025-01-06:
I've got the rework done mostly - and merged into Master. I still need to do the control classes, but I have some refactoring I want to do there at the same time.  And I need to `classify` `system_discovery` as well, but that isn't as urgent.

mitra42, 2025-01-07:
I've added a few things suggested above (or in messages I can't find) as issues
#106 
#107

mitra42, 2025-01-16:
- [ ] Move hysterisis control to control_hysterisis (from frugal_iot)
- [ ] Check the function call part of controls is working in response to  input
- [ ] maybe comment out the check for changed and hten follow code logic
- [ ] Make sure UI can handle float inputs and has a number field option instead of slider
- [ ] Expand ot have multiple types e.g. INfloat INuint16 etc
- [ ] Replace existing demo_mqtt
- [ ] Add timing
- [ ] Replace blinken
- [ ] Explore using a template for IN
- [ ] Auto discovery
- [ ] Use the INfloat etc types in other Sensor & Actuator

mitra42, 2025-05-02:
Now closing this - continuation in
- #139
- #106

closed 2025-05-02 by mitra42