mitra42/frugal-iot #44: WiFi configuration: pick up GPS

mitra42, 2024-11-03:
During WiFi configuration, it could try and get GPS from the user,  rational.

* This is going to be location specific, since will configure on-site knowing hte WiFi available. 
* Devices are unlikely to be moved - they'd need reconfiguring to new WiFi 
* Users can decline providing location if prefer for privacy reasons

mitra42, 2024-11-30:
Unfortunately - depends on https - so I can't see a way to use it when connected to WiFi portal
https://developer.mozilla.org/en-US/docs/Web/API/Geolocation_API

mitra42, 2024-12-05:
Idea .... on portal page after submit and before restart - "Set location" link goes to somewhere like
naturalinnovation.org:8080/gps?clientid=ESP32-abfeg 

* Security issue - maybe share a secret at the same time
  * Maybe .... link includes a key - valid for 5 mins - and esp32-abfeg when it gets connection sends something the server (not broker) including the key. 
  *

jksemple, 2024-12-28:
Just get the user to find their location on Google Maps, right-click and copy, then paste their lat/long into the device's portal page.  Suggest you restrict the Captive Portal to just getting the right WiFi details and handle all other config including location through a more comprehensive config website when the device has rebooted and connected.

mitra42, 2024-12-28:
That would work, however its illustrative of the design constraints of working in Resource Constrained environments - in particular the education and skill level of the users.  For example the pilot project will need sensors installed by people visiting sites - delivery people on minimum, Indonesian wage doing this alongside pickup of food waste. I believe means most will not have completed high school.  

One crucial design principle is that the more steps required the less likely to succeed. 

Connecting to WiFi and Password is something people are used to doing - the only steps on top of that required are to name the project its part of and the device name. 

My initial thinking about control flow - which may have flaws - was that the SAVE function redirects the mobile browser to a fixed location along with a device id, and some kind of secret to avoid spoofing.  The problem with that, is that the mobile browser is at this point connected to the portal's access point which has just been turned off. So the flow needs some rethinking.

jksemple, 2024-12-29:
It would be helpful to elaborate on the required accuracy for location. If devices need to be plottable on a farm level map so measured values can be correlated against location in greenhouses or fields the required accuracy will probably be higher than what can be obtained by interrogating the installers phone. If devices only need to be located to say the nearest km that can probably be done in bulk for all the devices in each installation

jksemple, 2024-12-29:
Also consider that location will inevitably need to be updated either following errors at installation time or because device has been moved so the ui function to set location should be accessible when the device has connected to WiFi as STA and not just in captive portal AP mode (which is troublesome to access in my experience)

mitra42, 2024-12-29:
I think applications will vary, the most common case I see is where a "project" is distributed across multiple physical sites and the geolocation is needed to put on a map.   Presuming low budget we need something that levers off the installer's phone. (By the way, the obvious idea of having the portal collect the GPS location doesn't appear to work because the browsers don't allow it for http and can't do https (with certificate) in portal mode. 

Second set of applications might be for moving objects, if the location is important enough to justify a few dollars, then can probably add a physical GPS. I've got one of these on order to try - https://www.aliexpress.com/item/1005004862529455.html (USD3.33+0.70 shipping).  This probably also handles those with high accuracy requirement.

I'm not going to worry about lowest budget / high-accuracy or moving yet.

jksemple, 2024-12-29:
I assume when you say 'portal' you mean when the device is operating in wifi AP mode as a captive portal. In that mode I agree you won't be able to access the Internet and https is a non starter. That's partly why I recommend just doing config of WiFi in the captive portal mode. After rebooting in WiFi STA mode you can then run a full-fat config website on the device which can make https calls in the background as necessary. My devices use https to check for OTA updates and to connect to Google Cloud services to dump SD data files. If you call anything synchronously that is somewhat time consuming you will find your http request triggers a watchdog exception. I found running such tasks (eg getting a GPS location or reading/writing files) was best done in another Free RTOS thread.

mitra42, 2024-12-30:
Thanks, A couple of points .... 
* If the devices move (as opposed to being mobile devices), they'll need to re-enter WiFi details, in which case that's also a time to enter new GPS 
* I'm not using Free RTOS explicitly - i.e. this is just the Arduino IDE main thread, partly to support other chips - most of the physical devices I'm playing with are ESP8266 which are quite a bit more affordable, and I'd like to support other things - though there may not be a need especially since there aren't AFAIK other cheap devices with WiFi. 
* I'm not sure how portable Free RTOS is - and its possible that one or more of the drivers I'm using is also inextricably tied to it (especially the MQTT) so it might be being incorporated indirectly. 
* I started using the expressif environment, but had some issues that suggested it was ESP32 only, and I definitely want to support the ESP8266s. 
*

jksemple, 2024-12-30:
Freertos is built into the ESP Arduino core already. Wifi runs on a separate freertos thread so it's there anyway whether you need to use it or not. Espasyncwebserver also runs on its own thread. Freertos is there if you need to run lengthy tasks without blocking the main loop().
I wrote a JobQueues library that allows you to queue up lambda functions to run in a background thread. It might be helpful if you find you can't call external services without impacting on sensor measurements or if you get watchdog idle timeout exceptions. 
The mqtt library that I use is knolleary / pubsubclient and that doesn't run in a separate thread but I imagine others may do. 
I've not tried using the esp-idf but I do now use vs code and platformio instead of arduino ide and its so much better. I'm not aware of any esp8266 limitations with platformio.

closed 2026-10-18 by mitra42