mitra42/frugal-iot #94: C++ Expert wanted ...

mitra42, 2024-12-16:
There are a few challenges that an expert C++ coder could help with .

- [ ] In actuator_digital.cpp - I'd like to pass a callback that is a method of an instance of a class, but I'm completely failing to get something that will compile. 
- [ ]

jksemple, 2024-12-28:
If you subclass all your different actuators from a common super Actuator which has a virtual inputReceived() method you can then pass across a reference to the actuator object instead of passing the callback.  That can then be held in your subscriptions collection and then the MQTT object can call actuator.inputReceived() on each such subscribed actuator object.  The appropriate concrete inputReceived() method will then get executed. I think that might work.

mitra42, 2024-12-28:
Yes - it might, the challenge is that about half of the sensors/actuators/control are not done as Classes - they are using the "setup/loop" paradigm of Arduino IDE.   Its a gradual rewrite to turn them into classes.

jksemple, 2024-12-28:
If you don't want to rewrite all your actuators as classes you'll just have to stick to the callback approach and where you have a class you'll need to define a callback function outside the class that calls this->inputReceived(). Not pretty, but you can't pass an object method as a function pointer parameter AFAIK.

mitra42, 2024-12-30:
Ok - its quite possibly the time to bite the bullet and rewrite as classes, not just for this reason.  I've got a issue for this somewhere and will stick a proposed class hierarchy there after the break. #25

jksemple, 2025-01-10:
In sensor.cpp you wanted to declare sensors as a static member of Sensor.  You can do this but you do need to initialise the static member before 'main' runs. Somewhere in sensor.cpp add the line std::vector<Sensor*> Sensor::sensors = std::vector<Sensor*>();

closed 2025-04-26 by mitra42