mitra42/frugal-iot #97: Build an adapter to demo the relay - box, Mains plug & socket,  maybe 5V power supply, ESP8266 Relay

mitra42, 2024-12-21:


jksemple, 2024-12-29:
Check out SONOFF switches: https://itead.cc/product/sonoff-basicr2/ 
These comprise an ESP8266 with a 10A mains relay and power supply.  They can be reprogrammed as a normal ESP8266 using Arduino IDE or VSCode/PlatformIO.  I use them to operate my pond pump, patio lights and a flood light from my MQTT home network.
They are available very cheaply from AliExpress.

mitra42, 2024-12-29:
Did you find programming them easily ? I checked them out a while ago, memory not certain but I seem to remember it involved quite a process to setup a programmer of some kind ?

jksemple, 2024-12-29:
You need to solder a 0.1" header to the sonoff board so you can connect a standard FTDI programmer but once you've burned in an OTA bootstrap that way you can flash new firmware over OTA. It's basically no different to programming a bare esp8266.

mitra42, 2024-12-30:
Maybe that was the point - need to get a FTDI programmer etc and will have to look at the soldering process, as that would need to be done in-country (will also be needed on the ESP8266/ESP32 boards and shields).  Its about the right price point though.

mitra42, 2025-03-17:
Sonoff now working - Got FTDI programmer - built cable - programmed Sonoff - 

there was one tweak needed to platform.ini had to add 
`board_build.flash_mode = dout`

It sitting in dev/lotus ie. clicking the "Relay' button turns on and off a light in my office :-)

At prices around USD4.84 its hard to beat that with any combination of power-supply; relay; and safe enclosure. 
https://www.aliexpress.com/item/4001315073122.html?

Final step to document the programming process, its non-trivial but completely do-able.

mitra42, 2025-04-16:
Closing - sonooff working great

closed 2025-04-16 by mitra42