mitra42/frugal-iot #108: Hardware: Sonoff switch

mitra42, 2025-01-07:
@jksemple mentioned the Sonoff switch in #97 , which I've heard mentioned elsewhere, 
```
These comprise an ESP8266 with a 10A mains relay and power supply. They can be reprogrammed as a normal ESP8266 using Arduino IDE or VSCode/PlatformIO. I use them to operate my pond pump, patio lights and a flood light from my MQTT home network.
They are available very cheaply from AliExpress.
You need to solder a 0.1" header to the sonoff board so you can connect a standard FTDI programmer but once you've burned in an OTA bootstrap that way you can flash new firmware over OTA. It's basically no different to programming a bare esp8266.
```

Example on a store - https://itead.cc/product/sonoff-basicr2/

Jon -  If you have one, I’d be very interested in whether you could get the code running on it.  I’m guessing it would be some variation of the actuator_relay - maybe just that actuator with the pin configured based on a board type.

I see someone suggesting it doesn't need soldering, just holding pins in place (which I've done on different boards with success) - see https://community.home-assistant.io/t/do-you-still-have-to-solder-the-modern-revisions-of-sonoff-plugs-to-flash-them/73995

jksemple, 2025-01-08:
The older Sonoff switches used an ESP8266. The new R2 ones use an ESP8285 but these behave indistinguishably from ESP8266 and 8285 does not need to be handled differently in code at all.  Just choose device type ITEAD_SONOFF when compiling.

To programme I soldered a 4-pin female 0.1" header to the PCB and made up a short link cable to hook up to the FTDI programmer.  Once OTA firmware delivery is working properly a Sonoff will only need to be programmed this way once so soldering a header could probably be avoided in production. For now its no big deal:
[image]

The frugal-iot code runs without modification.  In _local.h I asserted ACTUATOR_RELAY_WANT and ANY_DEBUG and set ACTUATOR_RELAY_PIN to 12. I also provided my local WiFi credentials.  Then:

[image]

and using the MQTT Dash app configured to point to the naturalinnovation MQTT server I was able to toggle the relay state:
[image]

mitra42, 2025-01-08:
That's cool that it took only about 10 lines in the config file to support a new device like this.

mitra42, 2025-01-08:
Does the Sonoff have an LED or anything else that is under program control ? 

 I'm presuming at that low price it doesn't have any kind of voltage or current sensing.

jksemple, 2025-01-09:
The LED is pin 13. 

It has no other external sensors however the ADC can be configured to read the uC supply voltage https://docs.espressif.com/projects/esp8266-rtos-sdk/en/latest/api-reference/peripherals/adc.html which could have been useful to detect incipient brown out conditions if the device were not powered from the mains.  It is possible that the 3V3 rail is smoothed enough that a MQTT message could be sent out warning of imminent power failure as the voltage drops but it would be a long-shot.

mitra42, 2025-01-10:
Thanks - looks good, going to get hold of some of these and play with them. 
I merged your _local.h example changes into _local-template.h

closed 2025-01-10 by mitra42